nakujaproject/avionicsN2_JLC_Computer #1: Telemetry

Mike-Kimani, 2021-06-27:
Using LoRa

0x6f736f646f, 2022-01-08:
Done on [this](https://github.com/nakujaproject/N2-onboardTelemetry) repo

SammyOina, 2022-01-09:
Consider adding additional I2C lines for future components

SammyOina, 2022-01-09:
@0x6f736f646f there are incomplete nets such as LEDR, MOSI and others

0x6f736f646f, 2022-01-09:
> @0x6f736f646f there are incomplete nets such as LEDR, MOSI and others

For which design?
I was saying the issue shouldn't be on this repo as it doesn't have a telemetry PCB. I agree with you @Mike-Kimani should provide i2c lines for telemetry

SammyOina, 2022-01-09:
> > @0x6f736f646f there are incomplete nets such as LEDR, MOSI and others
> 
> For which design? I was saying the issue shouldn't be on this repo as it doesn't have a telemetry PCB. I agree with you @Mike-Kimani should provide i2c lines for telemetry

For the telemetry PCB, I can create an issue in that repo instead

0x6f736f646f, 2022-01-09:
Okay

Mike-Kimani, 2022-03-08:
I2C lines for telemetry provided.

closed 2022-03-08 by Mike-Kimani