nakujaproject/avionicsN2_JLC_Computer #10: Review PCB

Mike-Kimani, 2021-12-13:


0x6f736f646f, 2021-12-13:
@Mike-Kimani Provide I2C communication lines for sending data to the telemetry computer

0x6f736f646f, 2022-01-08:
> @Mike-Kimani Provide I2C communication lines for sending data to the telemetry computer

Feedback on this?

SammyOina, 2022-01-09:
Consider switching buzzer to active buzzer such as mlt-7525

0x6f736f646f, 2022-01-09:
Also nimeona some strange LED's why those?

Mike-Kimani, 2022-01-10:
@0x6f736f646f  I'll provide the I2C lines

Mike-Kimani, 2022-01-10:
> Also nimeona some strange LED's why those?

Cool factor.

0x6f736f646f, 2022-01-10:
> > Also nimeona some strange LED's why those?
> 
> Cool factor.

What is a cool factor?
Are they RGB? ama?

Mike-Kimani, 2022-01-25:
@SammyOina Comment on this.

SammyOina, 2022-02-15:
> @SammyOina Comment on this.

of course, RGB is cool. The ws2812b provides a wide range of indicators that can be used to demonstrate different states. They can also be paired up and addressed using a single data line

closed 2022-03-02 by 0x6f736f646f