nakujaproject/avionicsN2_JLC_Computer #13: Add overvoltage protection

0x6f736f646f, 2022-01-08:
Add over-voltage protection which shuts down the supply, or clamps the output, when the voltage exceeds a preset level.

Mike-Kimani, 2022-01-09:
What exact component do you want us to protect from over-voltage?

0x6f736f646f, 2022-01-09:
All the components.

0x6f736f646f, 2022-01-09:
Specifically on the power input the voltage regulators

Mike-Kimani, 2022-01-10:
We'll use specific battery voltage. There is hence no risk of having an overvoltage.

0x6f736f646f, 2022-01-10:
> We'll use specific battery voltage. There is hence no risk of having an overvoltage.

😂That is deciding to ignore the eminent threat.
What if some plugs in a different battery with a different voltage?

Mike-Kimani, 2022-01-10:
The Voltage regulator used allows for twice the amount of voltage that is on the schematic. I don't think we will have a problem with overvoltage.

0x6f736f646f, 2022-01-10:
Okay

Mike-Kimani, 2022-01-10:
Alright.

closed 2022-01-10 by Mike-Kimani