nakujaproject/avionicsN2_JLC_Computer #16: Decoupling capacitors

0x6f736f646f, 2022-01-08:
Place the decoupling capacitors as close as possible to an integrated circuit (IC) on a PCB layout.
Example c5 and c6 should be as close to the vin and vout respectively on u7

Ref
1. [Here](https://i.stack.imgur.com/ENvPJ.png)

Mike-Kimani, 2022-01-09:
I'm redesigning the circuit to move to a rectangular board. I'll do this then.

0x6f736f646f, 2022-01-09:
No problem. Why rectangular though?

Mike-Kimani, 2022-01-10:
Allows for more area as the height is adjustable.

0x6f736f646f, 2022-01-10:
> Allows for more area as the height is adjustable.

Do you have an image of how they will be stacked?

Mike-Kimani, 2022-01-10:
An avionics bay back to back for Avionics and flight control. The telemetry one could be circular and stacked above the two.

0x6f736f646f, 2022-02-16:
Decoupling capacitors should be placed near the +ve line, c12 and c13 should be near leds

Mike-Kimani, 2022-02-20:
They are.

closed 2022-03-02 by 0x6f736f646f