nakujaproject/avionicsN2_JLC_Computer #25: Connectors

0x6f736f646f, 2022-02-01:
Why are we using this connector, can't we use standard connectors?

[image]

Mike-Kimani, 2022-02-10:
It faces downwards allowing for easier connection of wires to the telemetry PCB. Could you expound on what you mean by a standard connector?

0x6f736f646f, 2022-02-10:
Can't we just just the normal male header pins then we can have them either straight or right angled

Mike-Kimani, 2022-02-13:
We can have them angled. Either male or female.

0x6f736f646f, 2022-02-16:
Why is H4 at the middle?

0x6f736f646f, 2022-02-16:
What is H2 for?

0x6f736f646f, 2022-02-16:
Add 12v connectors for the camera

0x6f736f646f, 2022-02-16:
Add 5v and 3.3v connector for telemetry

Mike-Kimani, 2022-02-20:
> Why is H4 at the middle?

H4 is just a jumper to complete battery power to the board. It's position doesn't really matter.

Mike-Kimani, 2022-02-20:
> What is H2 for?

It's a feature that allows for delay when setting up on the rocket.

Mike-Kimani, 2022-02-20:
> Add 5v and 3.3v connector for telemetry

+3.3V already added. You need 5V too? What for?

0x6f736f646f, 2022-02-20:
> > Add 5v and 3.3v connector for telemetry
> 
> +3.3V already added. You need 5V too? What for?

For WS2812

0x6f736f646f, 2022-02-20:
> > What is H2 for?
> 
> It's a feature that allows for delay when setting up on the rocket.

Why can't this delay be done in the software as we would have more than 1 minute between arming the avionics and firing ignition

0x6f736f646f, 2022-02-20:
> > Why is H4 at the middle?
> 
> H4 is just a jumper to complete battery power to the board as explained in the other issue. It's position doesn't really matter.

Then why is it at the centre and not at the periphery?

Mike-Kimani, 2022-02-23:
Power goes from the screw terminal to the jumper and then to the rest of the circuit. Do you have a technical reason for having it at the periphery? If so, kindly post it here.

Mike-Kimani, 2022-02-25:
> > > What is H2 for?
> > 
> > 
> > It's a feature that allows for delay when setting up on the rocket.
> 
> Why can't this delay be done in the software as we would have more than 1 minute between arming the avionics and firing ignition

I still want us to have a testing mode that does not have the delay. This will be done using both software and hardware.

0x6f736f646f, 2022-03-01:
> I still want us to have a testing mode that does not have the delay. This will be done using both software and hardware.

Okay

closed 2022-03-02 by 0x6f736f646f