nakujaproject/avionicsN2_JLC_Computer #34: Change BMP sensor

0x6f736f646f, 2022-03-11:
It seems too expensive at $9 per piece

Mike-Kimani, 2022-03-16:
@0x6f736f646f 

For such issues kindly suggest a better alternative from a technical point of view to the one that exists on the board.

jamesmacharia861, 2022-03-16:
found a cheaper pressure sensor that uses an accelerometer to detect motion states
[1811031937_STMicroelectronics-LIS3DHTR_C15134.pdf](https://github.com/nakujaproject/avionicsv2/files/8259251/1811031937_STMicroelectronics-LIS3DHTR_C15134.pdf)

0x6f736f646f, 2022-03-16:
> found a cheaper pressure sensor that uses an accelerometer to detect motion states [1811031937_STMicroelectronics-LIS3DHTR_C15134.pdf](https://github.com/nakujaproject/avionicsv2/files/8259251/1811031937_STMicroelectronics-LIS3DHTR_C15134.pdf)

@jamesmacharia861 what you have refered to is an accelerometer but this issue we can to change the pressure sensors to obtain altitude data

jamesmacharia861, 2022-03-22:
DSP310 aka C130156 going at   3.4840 dollars. Its applied in drones and has a favourable voltage consumption of1.2- 3.6V VDD and a low current requirement
[DSP310.pdf](https://github.com/nakujaproject/avionicsv2/files/8321117/DSP310.pdf)

closed 2022-05-03 by Mike-Kimani