nakujaproject/avionicsN2_JLC_Computer #40: Capacitor on GPS

0x6f736f646f, 2022-03-13:
The capacitor on the GPS should be in series or parallel?
Because I am assuming you are doing some sort of impedance matching or?

Mike-Kimani, 2022-04-15:
From the datasheet.

[image]

0x6f736f646f, 2022-04-15:
The capacitor around the antenna

0x6f736f646f, 2022-04-15:
Is it a matching circuit or a filtering circuit?

Mike-Kimani, 2022-04-19:
Should be matching. Issue also discussed [here.](https://github.com/nakujaproject/avionicsv2/issues/43)

0x6f736f646f, 2022-04-19:
If it is a matching circuit then the cap should be in parallel

Mike-Kimani, 2022-04-20:
The datasheet says otherwise.

Mike-Kimani, 2022-04-20:
@georgebangeboss Thoughts on this?

Mike-Kimani, 2022-04-20:
> Should be matching. Issue also discussed [here.](https://github.com/nakujaproject/avionicsv2/issues/43)

It's not a matching circuit. I was wrong on this. It is a series band pass filter.

georgebangeboss, 2022-04-20:
The capacitor is a DC blocking capacitor
The LNA AT2659 datasheet provides sufficient information about the matching circuits. The datasheet is at [https://github.com/nakujaproject/avionicsv2/tree/reviews2/GPS](url)

Here are photo snippets
[image]

[image]

closed 2022-05-03 by Mike-Kimani