nakujaproject/avionicsN2_JLC_Computer #43: Review GPS Module

georgebangeboss, 2022-04-07:


georgebangeboss, 2022-04-08:
The impedance matching circuits are perfect. What is left is adjusting the  widths of RF tracks to achieve controlled track impedance of 50 Ohms. Is the PCB coplanar or microstrip? What substrate material is being used? FR4? Those are used to calculate the correct track width

The 10 uF Bypass capacitor should be as close as possible to the VCC pin of LNA(AT2659) which seems okay visually

Will the shipped PCB come with the GPS antenna?

Mike-Kimani, 2022-04-09:
The PCB is microstrip.

As for the substrate:
[image]

As per now we'll fabricate the antenna ourselves.

georgebangeboss, 2022-04-11:
L1_1 and RF nets should be 0.29337mm trace width to achieve 50 Ohm controlled impedance

11.55 mil is equivalent to 0.29337mm

[image]

closed 2022-05-03 by Mike-Kimani